Azure/iot-edge-opc-publisher-v1 #294: Data ingestion takes 10 min.  How to speed up or what we are observing is normal?

subbu-ai, 2021-01-18:
Please find attached the log file.  I added more and more tags to OPCPublisher_1 container.  There is only single OPCPublisher we are streaming data.  The log file shows 25 min as  ingestion duration.  What are the implications, can we speed up, or this is normal.  We are using AMQP TCP.  During 25 min the data is continuously going to the IoTHub or it is waiting for the 25 min block to complete?
[OPCPublisher_1_20210118084700.log](https://github.com/Azure/iot-edge-opc-publisher/files/5831764/OPCPublisher_1_20210118084700.log)

karok2m, 2021-01-21:
@subbu-ai, can you please provide `published_nodes.json` files that you used ? And, which version of the publisher are you using ?

barnstee, 2021-01-25:
To answer your questions: OPC Publisher will send data as fast as your Internet connection allows. To speed up, you will need a faster Internet connection, where faster mainly refers to latency, not throughput. OPC Publisher will also start sending data as soon as it starts processing publishing files (publishednodes.json) or publishing jobs from the cloud. As Karapet mentions, to tell you if the perf is what to expect, please provide the publishednodes.json file.

barnstee, 2021-01-28:
Closing due to inactivity.

closed 2021-01-28 by barnstee