Azure/iot-edge-opc-publisher-v1 #295: Modifying node display name / Publish node description

fijo1015, 2021-01-19:
Hi,

I want to change the display name and/or the description of a node using a method. UAExpert shows the modified names, but the display name in Azure is still the old one. I do not get an error message.

1. Doesn´t OPC Publisher support the modification of the display name? 
2. And is it possible to publish the node description as well? 

I am using version 2.5.3. 

Best regards 
jo

barnstee, 2021-01-19:
Changing the display name in UA Expert doesn't change it on the server, so what OPC Publisher "sees" stays the same. We do however have the ability to specify a display name in the published nodes JSON, please check the docs on how this can be done.

fijo1015, 2021-01-27:
I am using the following writeDescription method:
> UA_Server_writeDescription(UA_Server *server, const UA_NodeId nodeId, const UA_LocalizedText description)

When I am reading out the description of the same node, then I get the changed node description. Therefore I am using the following method:
> UA_Server_readDescription(UA_Server *server, const UA_NodeId nodeId, UA_LocalizedText *outDescription)

The same procedure can be done using the DisplayName. Are you sure the NodeDescription and NodeDisplayName doesn´t change on the server, when using this methods?

barnstee, 2021-01-27:
That really depends on the implementation on the server. If there is a method to update the description/display name, then it is possible, yes.

closed 2021-01-28 by fijo1015