Azure/iot-edge-opc-publisher-v1 #301: Cryptic string in json-file

fijo1015, 2021-02-12:
Hi,

i am using OPC Publisher in combination with the message routing of the IoT Hub. I want to store the incoming messages in a json-file (Blob Storage). When i am doing this, the body of the file is a cryptic string. When i am storing the incoming messages in an avro-file, it works very well. Can anybody tell me, what is the reason for that behavior? What can I do to "see" the data in the json-file as well?

Best regards
jo
[image]

fijo1015, 2021-02-19:
It´s the issue #267. I am closing this issue.

closed 2021-02-19 by fijo1015

faroqahmd, 2021-02-24:
cryptic message is apparently base64 encoded data. You can use an online base64 to text decoders to observe data