Azure/iot-edge-opc-publisher-v1 #302: Error: Opc.Ua.ServiceResultException: Cannot access certificate private key

ksaye, 2021-02-23:
OPC Publisher Version:  2.7.206.36449
OS: Ubuntu 18.04
hostname of iotedge: home
Module URI:  mcr.microsoft.com/iotedge/opc-publisher:latest
Container Create Options:

```
{
    "Hostname": "publisher",
    "Cmd": [
        "publisher",
        "--pf=/appdata/publishednodes.json",
        "--aa",
        "--diagnosticsinterval=120",
        "--tm",
        "--lf=/appdata/publisher.log",
        "--ll=debug",
        "PkiRootPath=/appdata/pki",
        "ApplicationName=opcpublisher",
        "ApplicationCertificateSubjectName=CN=opcpublisher,O=device1,DC=home"
    ],
    "HostConfig": {
        "Binds": [
            "/iiotedge:/appdata",
            "/iiotedgecerts:/appdata/pki"
        ]
    }
}
```

Commands to create and copy the certificates:
```
openssl req -x509 -sha256 -nodes -days 3650 -newkey rsa:2048 -subj "/CN=opcpublisher/O=device1/DC=home" -keyout privateKey.key -out publicCert.crt
openssl pkcs12 -export -keypbe NONE -certpbe NONE -out privateKey.pfx -inkey privateKey.key -in publicCert.crt
# just hitting enter when asked for the private key
openssl x509 -outform der -in publicCert.crt -out publicCert.der

cp publicCert.der /iiotedgecerts/own/certs/
cp publicCert.der /iiotedgecerts/trusted/certs/
cp privateKey.pfx /iiotedgecerts/own/private/
```

Permissions on the private key:
```
root@home:~/blog# ls -all /iiotedgecerts/own/private/
total 12
drwxr-xr-x 2 root root 4096 Feb 23 15:42 .
drwxr-xr-x 4 root root 4096 Feb 18 16:55 ..
-rwxrwxrwx 1  777 root 2469 Feb 23 16:10 privateKey.pfx
```

Error message:
```
[22:12:16 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Connecting all clients to home using MqttOverTcp.
[22:12:16 INF Root] Starting module OpcPublisher version 2.7.206.36449.
[22:12:16 INF Root] Initiating prometheus at port 9702/metrics
[22:12:16 INF Root] Prometheus metric server started.
[22:12:16 DBG Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Starting Module Host...
[22:12:16 WRN Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Bypassing certificate validation for client.
[22:12:16 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Running in iotedge context.
[22:12:16 DBG Microsoft.Azure.Devices.Device.Client] [Info] Informational: IotHubConnectionString#63993496 .ctor IAuthenticationMethod is AuthenticationWithTokenRefresh: ModuleAuthenticationWithHsm#36936550
[22:12:16 DBG Microsoft.Azure.Devices.Device.Client] [Info] Informational: PipelineContext#53180767 Set ITransportSettings[] = Microsoft.Azure.Devices.Client.ITransportSettings[1]
[22:12:16 DBG Microsoft.Azure.Devices.Device.Client] [Info] Informational: PipelineContext#53180767 Set IotHubConnectionString = IotHubConnectionString#63993496
[22:12:16 DBG Microsoft.Azure.Devices.Device.Client] [Info] Informational: PipelineContext#53180767 Set OnMethodCalledDelegate = OnMethodCalledDelegate#358618857
[22:12:16 DBG Microsoft.Azure.Devices.Device.Client] [Info] Informational: PipelineContext#53180767 Set Action`1 = System.Action`1[Microsoft.Azure.Devices.Shared.TwinCollection]
[22:12:16 DBG Microsoft.Azure.Devices.Device.Client] [Info] Informational: PipelineContext#53180767 Set ConnectionStatusChangesHandler = ConnectionStatusChangesHandler#362428870
[22:12:16 DBG Microsoft.Azure.Devices.Device.Client] [Info] Informational: PipelineContext#53180767 Set OnReceiveEventMessageCalledDelegate = OnReceiveEventMessageCalledDelegate#396718990
[22:12:16 DBG Microsoft.Azure.Devices.Device.Client] [Info] Informational: PipelineContext#53180767 Set ProductInfo = .NET/1.26.0 (.NET Core 3.1.10; Linux 4.15.0-135-generic #139-Ubuntu SMP Mon Jan 18 17:38:24 UTC 2021; X64)
[22:12:16 DBG Microsoft.Azure.Devices.Device.Client] [Info] Informational: PipelineContext#53180767 Set ClientOptions = (null)
[22:12:16 DBG Microsoft.Azure.Devices.Device.Client] [CreateFromConnectionString] Informational: InternalClient#10598606 HostName=home;DeviceId=opc;ModuleId=opcpublisher Mqtt_Tcp_Only
[... 60 lines cut ...]
   at Autofac.ResolutionExtensions.TryResolveService(IComponentContext context, Service service, IEnumerable`1 parameters, Object& instance)
   at Autofac.ResolutionExtensions.ResolveService(IComponentContext context, Service service, IEnumerable`1 parameters)
   at Autofac.ResolutionExtensions.Resolve(IComponentContext context, Type serviceType, IEnumerable`1 parameters)
   at Autofac.ResolutionExtensions.Resolve[TService](IComponentContext context, IEnumerable`1 parameters)
   at Microsoft.Azure.IIoT.Modules.OpcUa.Publisher.ModuleProcess.RunAsync() in D:\a\1\s\modules\src\Microsoft.Azure.IIoT.Modules.OpcUa.Publisher\src\ModuleProcess.cs:line 111
[22:12:32 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Stopping worker supervisor
[22:12:32 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Stopping worker...
```

If I delete my certs and let the opcpublisher create them, it works fine as shown by the tree below.  My goal is to use specified certs with a date longer than one year. 

```
root@home:/iiotedgecerts# tree
.
├── own
│   ├── certs
│   │   └── opcpublisher [652D9F5BDD9FDA935DD3520F3EADD123DB6CF344].der
│   └── private
│       └── opcpublisher [652D9F5BDD9FDA935DD3520F3EADD123DB6CF344].pfx
└── trusted
    └── certs
        └── opcpublisher [652D9F5BDD9FDA935DD3520F3EADD123DB6CF344].der

5 directories, 3 files
```

cristipogacean, 2021-02-24:
@ksaye, you can have a look on this one: https://github.com/Azure/Industrial-IoT/issues/757

ksaye, 2021-02-24:
I have been basing most of my work on that issue.  Let me add the "-extfile cert.ext" piece and retest.  I would be surprised if the extension issue caused the ```Cannot access certificate private key``` error message.  Will update once I complete the testing.

ksaye, 2021-02-24:
@cristipogacean, thanks for the comment.  The issue was based on missing extensions, and I have documented the process with the correct extensions here: https://kevinsaye.wordpress.com/2021/02/24/using-custom-x509-certificates-in-iot-edges-opc-publisher/.  Closing this ticket.

closed 2021-02-24 by ksaye