Azure/iot-edge-opc-publisher-v1 #305: IoTCentral flag not working as expected

zkaiser-ff, 2021-04-05:
I'm using the OPC Publisher module on a linux machine to read tags from an OPC UA server with the intent of publishing them to an IoT Central application. I've used the `--ic` flag and am still getting data unmodeled in IoT Central. I've had no problems connecting to the OPC UA server and tags are publishing with correct values, just not in the format IoT Central requires.

Module config
```
"OPCPublisher": {
            "version": "1.0",
            "type": "docker",
            "status": "running",
            "restartPolicy": "always",
            "settings": {
              "image": "mcr.microsoft.com/iotedge/opc-publisher:latest",
              "createOptions": {
                "HostConfig": {
                  "Binds":[
                    "/iiotedge:/appdata"
                  ]
                },
                "Hostname": "publisher",
                "Cmd": [
                  "--pf=/appdata/pn.json",
                  "--ic=true",
                  "--di=60",
                  "--to",
                  "--aa",
                  "--si=120",
                  "--ms=0"                  
                ]
              }
            }
          }
```

pn.json
```
[
  {
    "EndpointUrl": "opc.tcp://{ipaddress}:62541/discovery",
    "UseSecurity": false,
    "OpcAuthenticationMode": "Anonymous",
    "OpcNodes": [
      {
        "Id": "ns=2;s=[IoT]/Barn 602/External Temperature",
        "DisplayName": "externaltemperature",
        "OpcSamplingInterval": 1000,
        "OpcPublishingInterval": 5000
      },
      {
        "Id": "ns=2;s=[IoT]/Barn 602/House Temperature",
        "DisplayName": "housetemperature",
        "OpcSamplingInterval": 1000,
        "OpcPublishingInterval": 5000
      }
    ]
  }
]
```

Iot Central telemetry
[image]

My understanding was that by setting the `--ic=true` flag, that the data would be formatted with the display name in `pn.json` as the key of the payload which then can be pulled out in IoT Central.

Any insights or assistance would be appreciated.
Thanks!

koepalex, 2021-04-09:
The command line argument `--ic` is not supported in OPC Publisher > 2.6, from CLI documentation:

```csharp
"ic|iotcentral=", "Legacy - do not use." 
```
Maybe using PubSub format `--mm=PubSub` will do what you need:
```json
{
  "MessageId": "18",
  "MessageType": "ua-data",
  "PublisherId": "uat46f9f8f82fd5c1b42a7de31b5dc2c11ef418a62f",
  "DataSetClassId": "78c4e91c-82cb-444e-a8e0-6bbacc9a946d",
  "Messages": [
    {
      "DataSetWriterId": "uat46f9f8f82fd5c1b42a7de31b5dc2c11ef418a62f",
      "SequenceNumber": 18,
      "MetaDataVersion": {
        "MajorVersion": 1,
        "MinorVersion": 1
        },
        "Timestamp": "2020-03-24T23:30:56.9597112Z",
        "Status": null,
        "Payload": {
            "http://test.org/UA/Data/#i=10845": {
                "Value": 99,
                "SourceTimestamp": "2020-03-24T23:30:55.9891469Z",
                "ServerTimestamp": "2020-03-24T23:30:55.9891469Z"
              }
         }
    }
  ]
}
```

zkaiser-ff, 2021-04-09:
@koepalex 

Thanks for the information! Where are you seeing that documentation? I was looking at the following link for setting it up and that doesn't say anything about the `--ic` flag being legacy.

https://docs.microsoft.com/en-us/azure/iot-accelerators/howto-opc-publisher-run

koepalex, 2021-06-10:
As you see the whole documentation you linked is marked as outdated: 
[image]

Closing the issue.

closed 2021-06-10 by koepalex