Azure/iot-edge-opc-publisher-v1 #309: Cannot log into grafana dashboard

creationsoft, 2021-04-14:
I cannot log into the grafana dashboard, either when running in IoT Edge or a local docker instance.
I have tried changing permissions on the grafana.db file but cannot do this (always get 'operation not permitted')

INFO[04-14|12:20:59] Request Completed                        logger=context userId=0 orgId=0 uname= method=POST path=/login status=401 remote_addr=172.17.0.1 time_ms=10 size=42 referer=http://localhost:3000/login
EROR[04-14|12:21:01] Error while signing in user              logger=context userId=0 orgId=0 uname= error="failed to create auth token: attempt to write a readonly database" remote_addr=172.17.0.1
EROR[04-14|12:21:01] Request Completed                        logger=context userId=0 orgId=0 uname= method=POST path=/login status=500 remote_addr=172.17.0.1 time_ms=12 size=41 referer=http://localhost:3000/login

koepalex, 2021-06-07:
@creationsoft please follow the tutorial [Metrics Dashboard](https://github.com/Azure/iot-edge-opc-publisher/blob/main/docs/prometheus-metrics/MetricsDashboard.md) see `Update the environment variables to control access to the Grafana dashboard.`

koepalex, 2021-06-10:
Closing the issue; please reopen if there still open

closed 2021-06-10 by koepalex