Azure/iot-edge-opc-publisher-v1 #311: Container Create Options Continually Overwritten by Deployment Defaults

tmote, 2021-04-17:
I am new to publishing modules via IOT Edge and fully expect this could be a question for another level within the solution. But since this is only happening with the OPC Publisher module, I am starting here.

I have (re)deployed the OPC Publisher module multiple times to attempt to clear the issue. However, within five minutes of configuring the base "Container Create Options" to provide the storage bindings and necessary configuration to pick up the configuration from the JSON document, the settings are returned to the default.

They do take effect, as the local storage folder was created (as it did not exist). However, nothing else occurs. I have a "publishednodes.json" document in the folder on the host with a small configuration test. But I do not get any logs written to the folder or certificates generated.

oliverchristen, 2021-05-04:
You probably have an IoT Edge Deployment assigned to that device, do you? When you manually change the device configuration on the Azure portal, this change automatically gets overridden by the deployment. Either you change the edge deployment or you unassign the deployment and configure the device manually on the poral.

tmote, 2021-05-04:
I've tried both a deployment and a direct module push. Both do the same thing, within five minutes the settings are reverted. Unlike the Modbus module that I have been testing which allows me to deploy the module and then manage it normally (twin settings) in the same manner but the settings are static.

oliverchristen, 2021-05-04:
Perhaps there is another deployment which is assigned to this edge device with higher priority. You can check which one is assigned to the edge device in tab "Deployments and Configurations". I assume this is not related to this module but to the IoT hub itself.
If you change anything in the Modbus module twin, this change is not overridden after some minutes?

tmote, 2021-05-04:
That's correct - the Modbus module works as part of a deployment (layered and alone) exactly as expected. Initially I did have a layered deployment with Modbus and OPC and backed out of it as an initial troubleshooting effort. I then used a standalone deployment with OPC and the same issue persisted. I backed out of this approach and then proceeded with a direct push to the device using set module directly. In all cases, I'm updating the module's twin settings to push to the device. The push occurs, and then within five minutes backs out to default. I can verify the push happens due to a folder that is created on the device for binding physical storage.

I am unable to reproduce the error with any other module using the same configuration management. But I can reproduce the error across devices with the OPC module.

trm

oliverchristen, 2021-05-05:
That sounds weird. But anyway I'm quite sure that this is not related to the module itself as it's a functionality of IoT hub which handles module twins. So I recommend to create an issue on the Azure portal for that.

koepalex, 2021-06-07:
@tmote does the error still occur? Could you provide some information

* Container Create Options
* Used Container Versions
* logs from edgeAgent in that case

koepalex, 2021-06-10:
Closing for new; please reopen if error still exist

closed 2021-06-10 by koepalex