Azure/iot-edge-opc-publisher-v1 #316: The public/private key pair in the certficates do not match.

terrymandin, 2021-05-31:
## Expected Behavior
OPC Publisher moduel should start

## Current Behavior
OPC Publisher module fails with "The public/private key pair in the certficates do not match." error.  The OPC Publisher module installs fine when I install to an Azure VM, but fails when I install on the customer's PC.

## Steps to Reproduce
1. Install in a locked down environment with restricted network access and no public internet.
2. Install IoT Edge for Windows offline
3. Configure to use "Symmetric Key" attestation.
4. Install OPC Publisher offline via Docker "save" and "load" methods
5. Start IoT Edge
6. OPC Publisher module fails with the message, "The public/private key pair in the certficates do not match."

## Context (Environment)

### Device Information
* Host OS: Windows 10 Enterprise N Version 1809 (OS Build 17763.1879)
* Architecture: amd64 (Windows Azure VM)
* Container OS: Windows containers

### Runtime Versions
* aziot-edged: 1.1.1
* Edge Agent: 1.1.2-windows-amd64
* Edge Hub: 1.1.2-windows-amd64
* OPC Publisher: 2.5.0-windows-amd64
* Docker/Moby: 19.03.12+azure

## Logs
</details>
<summary>opcPublisher logs</summary>

```
IoTEdge detected.
[08:50:05 INF] Current directory is: C:\appdata
[08:50:05 INF] Log file is: publisher-publisher.log
[08:50:05 INF] Log level is: verbose
[08:50:05 INF] OPC Publisher V2.3.0 starting up...
[08:50:05 DBG] Informational version: V2.3.0
[08:50:05 INF] OPC UA monitored item notifications with one of the following 2 status codes will not generate telemetry events:
[08:50:05 INF] StatusCode: BadNoCommunication (dec: 2150694912, hex: 80310000)
[08:50:05 INF] StatusCode: BadWaitingForInitialData (dec: 2150760448, hex: 80320000)
[08:50:05 INF] opcstacktracemask set to: 0x7FFFFFFF
[08:50:05 INF] OperationTimeout set to 120000
[08:50:05 INF] OPC UA server base address: opc.tcp://publisher:62222/UA/Publisher
[08:50:05 INF] Security policy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256 with mode SignAndEncrypt added
[08:50:05 INF] Trusted Issuer store type is: Directory
[08:50:05 INF] Trusted Issuer Certificate store path is: pki/issuer
[08:50:05 INF] Trusted Peer Certificate store type is: Directory
[08:50:05 INF] Trusted Peer Certificate store path is: pki/trusted
[08:50:05 INF] Rejected certificate store type is: Directory
[08:50:05 INF] Rejected Certificate store path is: pki/rejected
[08:50:05 INF] Rejection of SHA1 signed certificates is disabled
[08:50:05 INF] Minimum certificate key size set to 1024
[08:50:05 INF] Application Certificate store type is: Directory
[08:50:05 INF] Application Certificate store path is: pki/own
[08:50:05 INF] Application Certificate subject name is: publisher
[08:50:05 WRN] WARNING: Automatically accepting certificates. This is a security risk.
[08:50:05 INF] No existing Application certificate found. Create a self-signed Application certificate valid from yesterday for 12 months,
[08:50:05 INF] with a 2048 bit key and 256 bit hash.
[08:50:07 FTL]    at CertificateFactory.CreateCertificateFromPKCS12(Byte[] rawData, String password)
   at CertificateFactory.CreateCertificateWithPrivateKey(X509Certificate certificate, String friendlyName, AsymmetricKeyParameter privateKey, SecureRandom random)
   at CertificateFactory.CreateCertificate(String storeType, String storePath, String password, String applicationUri, String applicationName, String subjectName, IList`1 domainNames, UInt16 keySize, DateTime startTime, UInt16 lifetimeInMonths, UInt16 hashSizeInBits, Boolean isCA, X509Certificate2 issuerCAKeyCert, Byte[] publicKey, Int32 pathLengthConstraint)
   at OpcPublisher.OpcApplicationConfiguration.InitApplicationSecurityAsync() in C:\app\modules\opc-publisher\opcpublisher\OpcApplicationConfigurationSecurity.cs:line 197
   at OpcPublisher.OpcApplicationConfiguration.ConfigureAsync() in C:\app\modules\opc-publisher\opcpublisher\OpcApplicationConfiguration.cs:line 172
   at OpcPublisher.Program.MainAsync(String[] args) in C:\app\modules\opc-publisher\opcpublisher\Program.cs:line 697
System.NotSupportedException: Creating X509Certificate from PKCS #12 store failed ---> System.Security.Cryptography.CryptographicException: The public/private key pair in the certficates do not match.
   at CertificateFactory.VerifyRSAKeyPair(X509Certificate2 certWithPublicKey, X509Certificate2 certWithPrivateKey, Boolean throwOnError)
   at CertificateFactory.CreateCertificateFromPKCS12(Byte[] rawData, String password)
   --- End of inner exception stack trace ---
   at CertificateFactory.CreateCertificateFromPKCS12(Byte[] rawData, String password)
   at CertificateFactory.CreateCertificateWithPrivateKey(X509Certificate certificate, String friendlyName, AsymmetricKeyParameter privateKey, SecureRandom random)
   at CertificateFactory.CreateCertificate(String storeType, String storePath, String password, String applicationUri, String applicationName, String subjectName, IList`1 domainNames, UInt16 keySize, DateTime startTime, UInt16 lifetimeInMonths, UInt16 hashSizeInBits, Boolean isCA, X509Certificate2 issuerCAKeyCert, Byte[] publicKey, Int32 pathLengthConstraint)
   at OpcPublisher.OpcApplicationConfiguration.InitApplicationSecurityAsync() in C:\app\modules\opc-publisher\opcpublisher\OpcApplicationConfigurationSecurity.cs:line 197
   at OpcPublisher.OpcApplicationConfiguration.ConfigureAsync() in C:\app\modules\opc-publisher\opcpublisher\OpcApplicationConfiguration.cs:line 172
   at OpcPublisher.Program.MainAsync(String[] args) in C:\app\modules\opc-publisher\opcpublisher\Program.cs:line 697
[08:50:07 FTL]    at CertificateFactory.VerifyRSAKeyPair(X509Certificate2 certWithPublicKey, X509Certificate2 certWithPrivateKey, Boolean throwOnError)
   at CertificateFactory.CreateCertificateFromPKCS12(Byte[] rawData, String password)
System.Security.Cryptography.CryptographicException: The public/private key pair in the certficates do not match.
   at CertificateFactory.VerifyRSAKeyPair(X509Certificate2 certWithPublicKey, X509Certificate2 certWithPrivateKey, Boolean throwOnError)
   at CertificateFactory.CreateCertificateFromPKCS12(Byte[] rawData, String password)
[08:50:07 FTL] Publisher exiting...
```

<details>

terrymandin, 2021-05-31:
Also tried setting to X509Store which did not work.  ie.

```
--at=X509Store
```

karok2m, 2021-06-02:
@terrymandin, can you please add IoT Edge deployment manifest with container creation options and any storage/networking options that it might have ?

koepalex, 2021-06-02:
@terrymandin were do you see the message `The public/private key pair in the certificates do not match` it isn't in the attached log file.

koepalex, 2021-06-02:
> Also tried setting to X509Store which did not work. ie.
> 
> ```
> --at=X509Store
> ```

Running as Linux Docker container, you can map the certificate stores to the host file system by using the Docker volume binding. Did you tested this?

If it is still failing you could generate the application X509 certificate and private key by your own and copy it into the binding folder:

* pki/own/certs X509 certificate as `*.der`
* pki/own/private X509 certificate private key as `*.pfx` (without password)

For testing: just install OPC Publisher in environment were it has connectivity and copy the certificates from this instance.

terrymandin, 2021-06-02:
Thanks for the help @karok2m and @koepalex.  

@karok2m, I have asked the customer for the deployment manifest and will attach it shortly.  

@koepalex, I think you are not seeing the message as "certificates" is misspelled in the log.  It is "certficates" in the log.  We did try mapping the certificates store to the local drive, but note that these are Windows containers, not Linux containers.  However, we did not see any certificate folders created.  Thanks for the guidance, we will try making our own certificates.

I will follow-up with the deployment manifest shortly

terrymandin, 2021-06-02:
The manifest is below.  Unfortunately I don't think we will be able to copy the certificates over as when we map the directories to our local drive, the certificate directories are not being created.

{
    "modulesContent": {
        "$edgeAgent": {
            "properties.desired": {
                "modules": {
                    "SimulatedTemperatureSensor": {
                        "settings": {
                            "image": "<ACRName>.azurecr.io/azureiotedge-simulated-temperature-sensor:1.1.2-windows-amd64",
                            "createOptions": ""
                        },
                        "type": "docker",
                        "version": "1.0",
                        "env": {
                            "MessageCount": {
                                "value": -1
                            }
                        },
                        "status": "running",
                        "restartPolicy": "always"
                    },
                    "OPCPublisher": {
                        "settings": {
                            "image": "<ACRName>.azurecr.io/iotedge/opc-publisher:2.5.0-windows-amd64",
                            "createOptions": "{\"Hostname\":\"publisher\",\"Cmd\":[\"--aa\",\"--to\",\"--di=60\",\"--ll=verbose\",\"--at=Directory\",\"--ap=/appdata/\",\"--tp=/appdata/\",\"--ip=/appdata/\"],\"HostConfig\":{\"Binds\":[\"c:/iiotedge:c:/appdata\"]}}"
                        },
                        "type": "docker",
                        "version": "1.0",
                        "status": "running",
                        "restartPolicy": "always"
                    }
                },
                "runtime": {
                    "settings": {
                        "minDockerVersion": "v1.25",
                        "registryCredentials": {
                            "EnbridgeACR": {
                                "address": "<ACRName>.azurecr.io",
                                "password": "<password>",
                                "username": "<username>"
                            }
                        }
                    },
                    "type": "docker"
                },
                "schemaVersion": "1.1",
                "systemModules": {
                    "edgeAgent": {
                        "settings": {
                            "image": "<ACRName>.azurecr.io/azureiotedge-agent:1.1.2-windows-amd64",
                            "createOptions": ""
                        },
                        "type": "docker",
                        "env": {
                            "UpstreamProtocol": {
                                "value": "MQTT"
                            }
                        }
                    },
                    "edgeHub": {
                        "settings": {
                            "image": "<ACRName>.azurecr.io/azureiotedge-hub:1.1.2-windows-amd64",
                            "createOptions": ""
                        },
                        "type": "docker",
                        "env": {
                            "UpstreamProtocol": {
                                "value": "MQTT"
                            }
                        },
                        "status": "running",
                        "restartPolicy": "always"
                    }
                }
            }
        },
        "$edgeHub": {
            "properties.desired": {
                "routes": {
                    "route": "FROM /messages/* INTO $upstream"
                },
                "schemaVersion": "1.1",
                "storeAndForwardConfiguration": {
                    "timeToLiveSecs": 7200
                }
            }
        },
        "SimulatedTemperatureSensor": {
            "properties.desired": {
                "SendData": true,
                "SendInterval": 30
            }
        },
        "OPCPublisher": {
            "properties.desired": {}
        }
    }
}

koepalex, 2021-06-02:
Please test with latest OPC Publisher `2.7.206` or at least `2.5.4` which is the latest release from 2.5.x.
Also this are windows container, not Linux. Therefor directory store will not work, the certificates need to be installed in system X509store.

terrymandin, 2021-06-04:
Upgrading to OPC Publisher ```2.7.206``` solved the issue.   

This customer needed to pull all containers from their own container registry.  Other than the tag names, it was not clear which container I should be pulling from mcr.microsoft.com for edgeAgent, edgeHub, or the OPC Publisher.  Perhaps I missed some documentation, but if not I recommend adding some guidance to our documentation.  

Thank you for the assistance @koepalex!  I will close this issue.

closed 2021-06-04 by terrymandin